BruceDevices/firmware #729: Add functionality to save multiple WiFi networks.

zcaliptium, 2025-01-24:
# Problem
I always take my M5StickC with myself, so I have different WiFi networks around.

For example:
1. At home - I have WiFi router and my home network.
2. Outside - I share WiFi from my smartphone.
3. At work -  I use office internal network.
4. At summerhouse - I have there own home network.
5. At John's house - My friend John Smith have own WiFi too.
6. etc. (we can continue that list infinitely)

And in current state it is painful always change credentials.

# Solution
There should be feature to save credentials for multiple WiFi networks, like address book. also It will be cool also make them labeled.

And maybe store them in config in a way like:
```json
  "wifi": [
    {
      "menuName": "home net",
      "creds": {
        "ssid": "Motherland"
        "pwd": "pass1234",
      }
    },
    {
      "menuName": "my phone",
      "creds": {
        "ssid": "Winterland AP"
        "pwd": "d3adb33f",
      }
    },
    {
      "menuName": "work net",
      "creds": {
        "ssid": "Wonderland Inc."
        "pwd": "12345678",
      }
    }
  ]
```

bmorcelli, 2025-01-24:
It already work that way..

You might be losing your bruce.conf installing other fw in place

zcaliptium, 2025-01-26:
It also would be cool to have "Share" option for any known WiFi network, same as it done on smartphones. That simply display QR code on screen.

Sure I know about "Others -> QR Code"... But it just has own list that has to be edited separately. That is not really usable at current state.

closed 2025-05-24 by bmorcelli