BruceDevices/firmware #731: Unable to use SD Card and CC1101 at the same time

AlexC-0806, 2025-01-25:
**Describe the bug:**
I am not able to use the SD card while having the CC1101 connected and viceversa.

**Steps To Reproduce:**

- Device used: 
 -- M5Stick CPlus 2
 -- SD card reader module
-- CC1101

**Expected behavior:**
I can use the RF functions and the SD card funtions without any issue
**Actual behavior:**
The board crashes when I try to Scan/Copy with CC1101

**Additional context / logs:**

```
Selected: SD Card
Files listed with: 13 files/folders found
Selected: Scan/copy
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 3,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "wifi": {},
  "irTx": 19,
  "irRx": 33,
  "rfTx": 32,
  "rfRx": 33,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ]
}[108771][I][config.cpp:195] saveFile(): config file written successfully
[108822][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[118375][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 164
[123373][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 0
[137320][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 255
Selected: Scan/copy
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 3,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "ledBright": 75,
  "ledColor": "0",
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "wifi": {},
  "irTx": 19,
  "irRx": 33,
  "rfTx": 32,
  "rfRx": 33,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "mifareKeys": [],
  "gpsBaudrate": 9600,
  "startupApp": "",
  "wigleBasicToken": "",
  "devMode": 0,
  "disabledMenus": [],
  "qrCodes": [
    {
      "menuName": "Bruce AP",
      "content": "WIFI:T:WPA;S:BruceNet;P:brucenet;;"
    },
    {
      "menuName": "Bruce Wiki",
      "content": "https://github.com/pr3y/Bruce/wiki"
    },
    {
      "menuName": "Bruce Site",
      "content": "https://bruce.computer"
    },
    {
      "menuName": "Rickroll",
      "content": "https://youtu.be/dQw4w9WgXcQ"
    }
  ]
}[141192][I][config.cpp:195] saveFile(): config file written successfully
[141253][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[141266][W][sd_diskio.cpp:180] sdCommand(): crc error
[141372][W][sd_diskio.cpp:180] sdCommand(): crc error
[141478][W][sd_diskio.cpp:186] sdCommand(): token error [13] 0x4
[141484][E][sd_diskio.cpp:624] ff_sd_status(): Check status failed
[141490][W][sd_diskio.cpp:174] sdCommand(): no token received
[141595][W][sd_diskio.cpp:174] sdCommand(): no token received
[141701][W][sd_diskio.cpp:174] sdCommand(): no token received
[141807][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x11
[141814][W][sd_diskio.cpp:174] sdCommand(): no token received
[141921][W][sd_diskio.cpp:174] sdCommand(): no token received
[142027][W][sd_diskio.cpp:174] sdCommand(): no token received
[142132][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[142138][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[142146][W][sd_diskio.cpp:174] sdCommand(): no token received
[142252][W][sd_diskio.cpp:174] sdCommand(): no token received
[142358][W][sd_diskio.cpp:174] sdCommand(): no token received
[142463][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[142469][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[142476][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/bruce.conf) failed
Error 4
Error 5
Error 5
Error 5
```

AlexC-0806, 2025-01-25:
#249  shows that the issue has been persisting for a long time, is there any possibility of a fix?

bmorcelli, 2025-01-26:
It is not "fixable" it is just not possible without modding and adding transistors and changing the code..

When using CC1101, take the SD card out and in again after each signal saving

AlexC-0806, 2025-01-26:
Okok understood, but why for example in CatHack firmware SD card and CC1101 can work simultaneously? I understand that CatHack is not open-source so it's not possible to verify how it works precisely, but still it could be possible to use the same approach

bmorcelli, 2025-01-29:
It can't, actually.. if you pay attention, after you save one signal it asks you to turn off and on again to physically unmount the SD SPI inner driver, than you can save again..

And if have made use of the SDCard, you need to do this procedure again..

We could have done it in bruce, but we would lose boot image, boot sound, save configs on SD Card, etc..

So as I said earlier.. to use SDCard, you must take the card out and put again, open RF, read/save the signal, take the card out, open Menu, put SD in, open RF, read/save again

closed 2025-01-29 by bmorcelli