BruceDevices/firmware #732: RF - Cannot transmit a frequency (868.95 MHz) with CC1101

Alphapwned, 2025-01-25:
**Describe the bug:**
I can receive a 868.95 MHz frequency and save it, but I can't transmit it

**Steps To Reproduce:**
Record the frequency of an electric shutter remote control to open or close the shutter (in 868.95MHz) and try to reproduce it.
- Device used: 
CYD : https://fr.aliexpress.com/item/1005004961285750.html?spm=a2g0o.order_list.order_list_main.70.dbaa5e5bai3eQV&gatewayAdapt=glo2fra
CC1101 : https://fr.aliexpress.com/item/1005006733128122.html?spm=a2g0o.order_list.order_list_main.22.dbaa5e5bai3eQV&gatewayAdapt=glo2fra

**Expected behavior:**
Open or close the electric shutter

**Actual behavior:**
When I try to transmit the frequency, nothing happens.

**Additional context / logs:**
Here's one of the signals I recorded and wanted to emulate :

Filetype: Bruce SubGhz File
Version 1
Frequency: 868950016
Preset: 0
Protocol: RAW
RAW_Data: 10009 -407 423 -406 828 -390 434 -813 405 -418 831 -810 833 -403 422 -813 833 -812 832 -812 831 -813 423 -405 831 -811 406 -419 830 -388 433 -390 433 -390 433 -815 831 -390 432 -391 433 -813 829 -816 830 -818 404 -420 403 -417 406 -417 405 -418 405 -416 405 -418 405 -416 836 -810 407 -417 407 -417 404 -418 406 -415 406 -418 406 -414 419 -405 830 -814 831 -388 435 -388 434 -813 406 -419 416 -405 831 -388 434 -387 434 -390 434 -388 434 -813 831 -411 419 -390 432 -391 433 -812 832 -813 405 -418 830 -822 831 -390 434 -813 406 -417 829 -391 431 -813 428 -405 832 -390 431 -390 433 -813 831 -813 831 -824 406 -418 829 -390 434 -387 434 -390 431 -390 432 -813 834 -813 405 -418 405 -416 405 -418 405 -417 405 -416 405 -419 418 -405 829 -812 406 -417 403 -416 406 -416 406 -416 403 -416 831 -814 405 -416 406 -416 403 -416 406 -415 406 -416 831 -5374 10011 -405 423 -406 829 -390 434 -813 405 -418 830 -813 830 -406 421 -813 831 -815 829 -815 831 -813 422 -405 831 -812 406 -418 829 -390 431 -393 431 -390 431 -818 831 -391 433 -388 433 -813 831 -816 828 -818 406 -417 405 -417 405 -418 405 -419 402 -418 405 -420 402 -418 833 -815 404 -417 406 -418 403 -420 403 -417 405 -418 405 -418 830 -813 831 -390 431 -390 433 -815 404 -418 405 -418 829 -390 433 -391 430 -391 431 -391 433 -813 831 -814 839 -390 433 -813 405 -420 828 -391 431 -813 830 -827 829 -390 434 -388 432 -390 433 -391 431 -815 829 -824 406 -418 829 -817 402 -417 832 -814 403 -420 425 -406 406 -418 831 -811 406 -417 829 -813 831 -403 421 -813 405 -419 404 -418 403 -419 406 -415 406 -418 404 -417 419 -405 404 -417 829 -390 431 -390 432 -390 431 -388 433 -388 434 -813 404 -417 404 -417 406 -417 828 -390 433 -389 431 -390 429 -5377 10012 -405 423 -405 831 -388 433 -815 403 -421 829 -813 831 -405 421 -814 830 -816 829 -815 829 -815 419 -408 829 -815 404 -417 831 -389 433 -389 433 -390 432 -817 831 -390 431 -392 431 -816 829 -815 830 -818 405 -419 402 -419 403 -419 403 -419 403 -419 403 -421 403 -419 833 -815 402 -420 403 -420 403 -418 405 -419 402 -418 405 -419 829 -815 829 -390 434 -390 431 -815 404 -417 406 -417 831 -391 430 -391 431 -391 433 -388 434 -813 831 -814 839 -390 431 -816 405 -418 828 -391 433 -815 829 -826 827 -391 433 -388 434 -390 431 -391 433 -814 830 -823 406 -418 829 -813 406 -418 830 -814 405 -419 423 -408 405 -417

bmorcelli, 2025-01-25:
This CC1101 antenna is optimized for 433Mhz frequency.. it will have low performance with frequencies lesser than 350 and over 446Mhz..

So if you want to use it with other frequency (and have good results) choose a CC1101 made for the other frequency range

https://a.aliexpress.com/_mrkD41v

Alphapwned, 2025-01-25:
That's what I thought.
Thank you for confirming :)

Alphapwned, 2025-01-28:
@bmorcelli Do you know how to connect the CC1101 868MHZ antenna you recommended to make it work with Bruce predatory? 
From what I can see in the documentation, LoRa is not supported by Bruce Predatory so I don't know if it's possible.

closed 2025-05-24 by bmorcelli