BruceDevices/firmware #735: 1.8.1 Update Sucks: Issues with Ports File Structure

biscuitaep, 2025-01-25:
In the **1.8.1** update, you changed the structure of the ports file, and it sucks. For example, the buttons don’t have debounce. I added it myself, and it worked, but now, pressing the back button quits the page. The old void functions were much better and easier to edit. Since I make my own PCB, I was forced to do my own porting, but I won’t be able to use this version. I hope you fix it.

bmorcelli, 2025-01-26:
And it will change again..

biscuitaep, 2025-01-26:
So you are recommending me to dont try to port :) When is the new version coming?

bmorcelli, 2025-01-26:
Idk what are the changes you need to use on your boards.. the structure will be almost the same, relying on the variables SelPress, NextPress, PrevaPress, EscPress, AnyKeyPress..

The keyboard has 4 modes of operation: 3 and 5 buttons, touchscreen and keyboard (Cardputer and T-Deck)

These things won't change, what will change is the way (or when) these variables will be updated... Instead of using a separate task, which is causing some problems, to read on the same loop task..

The we did now is way easier to port, because you need to make only device related functions, reducing from 10 to 5 functions..

bmorcelli, 2025-01-26:
If you need help on your port, send us more details, so I can help you with it

biscuitaep, 2025-01-26:
**This is my InputHandler code the problem is as you can see if I press up button (I have tri-button setup) it triggers both EscPress and PrevPress in 1.8 version the firmware checks it separately and this problem doesn't happen can you help me for this situation with my code i can't trigger esc and prev press separately.** @bmorcelli 

`
/*********************************************************************
** Function: InputHandler
** Handles the variables PrevPress, NextPress, SelPress, AnyKeyPress and EscPress
**********************************************************************/
#define DEBOUNCE_DELAY 50 // Debounce Delay (ms)

unsigned long lastPressTime = 0;

void InputHandler(void) {
    unsigned long currentMillis = millis();

    // Debouncing
    if ((digitalRead(UP_BTN) == LOW || digitalRead(SEL_BTN) == LOW || digitalRead(DW_BTN) == LOW) &&
        (currentMillis - lastPressTime > DEBOUNCE_DELAY)) {
        
        lastPressTime = currentMillis; // Update last press time
        
        if (!wakeUpScreen()) {
            AnyKeyPress = true;
        } else {
            goto END;
        }
        
        // Check buttons
        if (digitalRead(UP_BTN) == LOW) {
            PrevPress = true;
            
            //FIX: ALWAYS TRIGGERS ESCAPE
            //EscPress = true;
        }
        if (digitalRead(DW_BTN) == LOW) {
            NextPress = true;
        }
        if (digitalRead(SEL_BTN) == LOW) {
            SelPress = true;
        }
    }

    END:
    if (AnyKeyPress) {
        long tmp = millis();
        while ((millis() - tmp) < 200 &&
               (digitalRead(UP_BTN) == LOW || digitalRead(SEL_BTN) == LOW || digitalRead(DW_BTN) == LOW));
    }

    
}
`

biscuitaep, 2025-01-26:
Also I cant use keyboard nothing happens when I press to any button

bmorcelli, 2025-01-26:
> This is my InputHandler code the problem is as you can see if I press up button (I have tri-button setup) it triggers both EscPress and PrevPress in 1.8 version the firmware checks it separately and [...]

In the previous implementation, `bool checkEscPress()` usually had the same code as  `bool checkPrevPress()` , so it behaves the "same" way..

Now these variables are being reset each time the taskInputHandler runs, [here](https://github.com/pr3y/Bruce/blob/7a7744700f059c933398265508d825b8248ec806/src/main.cpp#L36-L51)

and the behavior is the same, so no problem to set them both to true at the same trigger

biscuitaep, 2025-01-26:
> In the previous implementation, `bool checkEscPress()` usually had the same code as  `bool checkPrevPress()` , so it behaves the "same" way..
>
> Now these variables are being reset each time the [...]

Weird if i put them in same trigger for example while selecting pin in IR Config menu if i press up button it exits instead of going up

biscuitaep, 2025-01-26:
I Don't know how but problem solved my first code that I sent is working now I don't know how

here is the debounced button code you can use @bmorcelli @pr3y thank you for help also I want to contribute to this project how can I? I can add 2x NRF24 code and better optimization to 2.4GHz Jammer

closed 2025-01-26 by biscuitaep