BruceDevices/firmware #738: RFID RC522 does not work with M5Stick CPlus2

AlexC-0806, 2025-01-26:
**Describe the bug:**
I'm trying to interface the RC522 RFID reader to my M5 Stick C Plus 2, but it's not found. I already checked cables and they are correct according to the [pinout](https://github.com/pr3y/Bruce/blob/main/media/connections/rc522_stick.jpg). I also set` M5 RFID2
` in the config menu. 

**Steps To Reproduce:**

- Device used: 
- M5Stick CPlus 2
- RC522

**Expected behavior:**
I can read tags with my board

**Actual behavior:**
When clicking on Read Tag, i get the error "RFID module not found!"

**Additional context / logs:**
```
Selected: Read tag
[ 71701][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
unsupported serial command: _
failed: _

```

IncursioHack, 2025-03-21:
Could you test again? Also, did you go to the RFID Modules and select your device option?

NoNameBuilder, 2025-03-23:
using here latest beta, still same issue.
i tried older, stable versions, same issue
using mini rfid rc522 (mfrc522) with rc522 pinout

AlexC-0806, 2025-03-26:
Apparently, we need to do [circuital modifications](https://forum.arduino.cc/t/rc522-rfid-rc522-switching-spi-to-uart-interface-or-i2c-possible/425741/5) if we want to use the RC522 in I2C mode, so maybe it could be useful to add an SPI mode in Bruce configuration

IncursioHack, 2025-03-26:
Have you tested this here?

https://github.com/pr3y/Bruce/blob/main/media/connections/rc522_stick.jpg

Pleroy77, 2025-06-17:
got the same pb