BruceDevices/firmware #745: how do i make a custom ssid with beacon spam

N0Bullets-code, 2025-01-29:
how should it end i tested it with .ino and it doensnt work

Lamnxzp, 2025-01-29:
@N0Bullets-code 
Create a `.txt` file where each line represents the name of a Wi-Fi network to be created, and use the file from "Custom SSIDs".

Example (`wifi_list.txt`):
```
Network_1
Network_3
Network_4
Network_5
Network_6
```

N0Bullets-code, 2025-01-29:
> [@N0Bullets-code](https://github.com/N0Bullets-code) Create a `.txt` file where each line represents the name of a Wi-Fi network to be created, and use the file from "Custom SSIDs".
> 
> Example (`wifi_list.txt`):
> 
> ```
> Network_1
> Network_2
> Network_3
> Network_4
> Network_5
> Network_6
> ```

thanks bro i was just to dumb to see it how it works thanks it works

closed 2025-01-29 by N0Bullets-code