BruceDevices/firmware #748: Refactor JavaScript interpreter

Tawank, 2025-01-29:
| This is a project issue to track progress |

Add functions from the flipper zero JavaScript https://github.com/jamisonderek/flipper-zero-tutorials/wiki/JavaScript

- Globals
  - [x] console
  - [x] delay
  - [ ] ffi_address (flipper zero internal API, not possible in Bruce)
  - [ ] parse_int
  - [x] print
  - [x] require
  - [ ] to_string
  - [ ] to_hex-string
  - [ ] to_lower_case
  - [ ] to_upper_case
  - [x] filepath
  - [x] dirpath
- badusb/badble
  - [ ] setup
  - [ ] quit
  - [ ] isConnected
  - [ ] press
  - [ ] hold
  - [ ] release
  - [ ] print
  - [ ] println
  - [ ] altPrint
  - [ ] altPrintln
- blebeacon
  - [ ] isActive
  - [ ] setConfig
  - [ ] setData
  - [ ] start
  - [ ] stop
  - [ ] keepAlive
- dialog
  - [ ] message
  - [ ] custom
  - [ ] pickFile
- flipper/device
  - [ ] getModel
  - [ ] getName
  - [ ] getBatteryCharge
- gpio
  - [ ] init
  - [ ] read
  - [ ] write
  - [ ] startAnalog(reference?: double) (reference voltage not possible on esp32, need to implement as no-op function)
  - [ ] stopAnalog (on esp32 there is no need to init analog, need to implement as no-op function)
  - [ ] readAnalog
- keyboard
  - [ ] setHeader
  - [ ] text
  - [ ] byte
- math
  - [ ] abs
  - [ ] acos
  - [ ] acosh
  - [ ] asin
  - [ ] asinh
  - [ ] atan
  - [ ] atan2
  - [ ] atanh
  - [ ] cbrt
  - [ ] ceil
  - [ ] clz32
  - [ ] cos
  - [ ] exp
  - [ ] floor
  - [ ] is_equal
  - [ ] log
  - [ ] max
  - [ ] min
  - [ ] pow
  - [ ] random
  - [ ] sign
  - [ ] sin
  - [ ] sqrt
  - [ ] trunc
  - [ ] PI
  - [ ] E
  - [ ] EPSILON
- notification
  - [ ] success
  - [ ] error
  - [ ] blink
- serial
  - [ ] setup
  - [ ] end
  - [ ] write
  - [ ] read
  - [ ] readln
  - [ ] readBytes
  - [ ] readAny
  - [ ] expect
- storage
  - [ ] read
  - [ ] write
  - [ ] append
  - [ ] copy
  - [ ] move
  - [ ] exists
  - [ ] mkdir
  - [ ] remove
  - [ ] virtualInit
  - [ ] virtualMount
  - [ ] virtualQuit
- subghz
  - [ ] end
  - [ ] getFrequency
  - [ ] getRssi
  - [ ] getState
  - [ ] isExternal
  - [ ] setFrequency
  - [ ] setIdle
  - [ ] setRx
  - [ ] setup
  - [ ] transmitFile
- textbox
  - [ ] addText
  - [ ] close
  - [ ] clearText
  - [ ] isOpen
  - [ ] setConfig
  - [ ] show
- usbdisk
  - [ ] createImage
  - [ ] start
  - [ ] stop
  - [ ] wasEjected
- vgm
  - [ ] getPitch
  - [ ] getRoll
  - [ ] getYaw
  - [ ] deltaYaw
- widget
  - [ ] addBox
  - [ ] addCircle
  - [ ] addDisc
  - [ ] addDot
  - [ ] addFrame
  - [ ] addGlyph
  - [ ] addIcon
  - [ ] addLine
  - [ ] addRbox
  - [ ] addRframe
  - [ ] addText
  - [ ] addXbm
  - [ ] close
  - [ ] loadImageXbm
  - [ ] remove
  - [ ] isOpen
  - [ ] show

Tawank, 2025-01-29:
Copy existing global functions to adequate require modules:
- ir
  - [ ] irRead
  - [ ] irReadRaw
- display
  - [ ] color
  - [ ] fillScreen
  - [ ] setTextColor
  - [ ] setTextSize
  - [ ] drawString
  - [ ] setCursor
  - [ ] print
  - [ ] println
  - [ ] drawPixel
  - [ ] drawLine
  - [ ] drawRect
  - [ ] drawFillRect
  - [ ] drawJpg
  - [ ] drawGif
  - [ ] gifOpen
  - [ ] width
  - [ ] height
- http
  - [ ] get
  - [ ] post
  - [ ] put
  - [ ] patch
  - [ ] delete
- wifi
  - [ ] wifiConnect
  - [ ] wifiConnectDialog
  - [ ] wifiDisconnect
  - [ ] wifiScan

eadmaster, 2025-01-30:
Instead of a full refactoring , you could simply add a compatibility layer while keeping the old functions.

Tawank, 2025-01-30:
I didn't removed existing functions old scripts are working normally.

Tawank, 2025-01-30:
I said refactor because in some cases I refratored some older functions to accept flipper zero syntax, 
but old syntax is working normally.

eadmaster, 2025-01-30:
You know the mantra: "never break userspace"   :-)

Tawank, 2025-03-01:
Progress so far: 
I've created docs for new API: https://github.com/Tawank/bruce-js-tooling/blob/master/packages/bruce-sdk/DOCS.md
Added new functions and fixed existing functions.

I created dino game, http browser for bruce and tried to run NES emulator (failed).
https://github.com/Tawank/bruce-js-tooling/tree/master/examples

To do:
- [ ] create test script to test all "new" API and "old" API. We need to be sure that old and new is working, and we will have testing so we see regress.
- [ ] give more examples in the docs