BruceDevices/firmware #751: wifi scan via interpreeter

Sonys9, 2025-01-29:
**Describe the bug:**
`

const AllPasswords = "12345678|123456789|11111111|22222222|33333333|44444444|55555555|66666666|77777777|88888888|99999999|00000000|87654321|987654321".split('|');

setTextSize(1);
var Ycoord = 20;

while(true){
    if(getPrevPress()){ break; }
    drawString("Scanning...", 5, 0);
    var Networks = wifiScan();
    delay(5000);
    
    drawString(String(Networks), 5, 0);
    for(var i = 0; i<Networks.length; i++) {
        drawString(Networks[i]["SSID"], 5, Ycoord);
        Ycoord+=20;
    }
    delay(3000);
    fillScreen(0);
    drawString("Retrying...", 5, 0);
    delay(2000);
    fillScreen(0);
}`
cant scan networks

**Steps To Reproduce:**
mmm idk
- Device used:  m5stickc plus2

**Expected behavior:**
get wifi netrowks

**Actual behavior:**
nothing found

**Additional context / logs:**
all empty

bmorcelli, 2025-02-01:
I believe it had been patched, and the fix available for testing in Beta

Sonys9, 2025-02-02:
fixed thanks

closed 2025-02-02 by Sonys9