BruceDevices/firmware #760: Bad BLE spam

Gosheto1234, 2025-02-01:
**Is your feature request related to a problem? Please describe.**
No , it's just request for the future 

**Describe the solution you'd like**
Bluetooth feature that is connected to the bluetooth spam , when you advertise something and someone clicks on connect it connects to your device and you can then start remote attacks like the Bad BLE option with script

**Describe alternatives you've considered**

**Additional context**

GabriWar, 2025-03-25:
probably not possible, since the spam uses many ""senders"" and when the pairing device sends the response bruce is already spamming with another mac and everything, also i dont think theyre really ""pairing"" packets, i think bruce sends advertising packets, that might make the process of paiting even harder

GabriWar, 2025-03-25:
ill be looking into this once the custom ble spam is merged tho

closed 2025-05-24 by bmorcelli

dapsvi, 2026-04-18:
I know this has been closed a year ago but I implemented it and will test it tomorrow.