BruceDevices/firmware #761: Rfid module not found

KalyanSingh05, 2025-02-01:
I have downloaded (burned) latest 1.8 version of bruce in m5 stickC plus 2. And trying to connect the rfid module (PN532 on I2C) as shown in different tutorials but when i run the program (rfid - read tag) of bruce the screen freezez for 15 seconds and then it displays error message "rfid module not found".

And this problem is not only with read module, but same happens when i select other options from rfid.

AlexC-0806, 2025-02-03:
It happens to [me with RC522](https://github.com/pr3y/Bruce/issues/738), too. I have no idea what could be the issue

KalyanSingh05, 2025-02-03:
> It happens to [me with RC522](https://github.com/pr3y/Bruce/issues/738), too. I have no idea what could be the issue

I think this issue is with newer versions.

Lamnxzp, 2025-02-26:
The DIP switch on the PN532 needs to be 1 ON and 2 OFF for I2C.

KalyanSingh05, 2025-02-26:
> The DIP switch on the PN532 needs to be 1 ON and 2 OFF for I2C.
> 

Still the issue is same

AlexC-0806, 2025-02-26:
> The DIP switch on the PN532 needs to be 1 ON and 2 OFF for I2C.

And for RC522? Are there any specific modifications to make on the module?

Lamnxzp, 2025-02-26:
@KalyanSingh05
I tested with Bruce 1.8.2 with a M5stick C Plus2, and it's working correctly.

Did you selected "PN532 on I2C" in the "RFID Module" setting?
Can you send a picture of your module connected to your M5stick C Plus2?

KalyanSingh05, 2025-02-26:
> @KalyanSingh05
> I tested with Bruce 1.8.2 with a M5stick C Plus2, and it's working correctly.
> 
> Did you selected "PN532 on I2C" in the "RFID Module" setting?
> Can you send a picture of your module connected to your M5stick C Plus2?

Yes, i had selected "PN532 on I2C"

Lamnxzp, 2025-02-26:
@KalyanSingh05 
There is a button "Add files"
[image]

KalyanSingh05, 2025-02-26:
@Lamnxzp 
You can see all the images

[image]

[image]

[image]

dzireee, 2025-03-01:
i have the same issue using SPI, still waiting for a solution

KalyanSingh05, 2025-03-01:
@Lamnxzp i have uploaded the images please have a look

Lamnxzp, 2025-03-01:
@KalyanSingh05 So.. the same module on Bruce 1.8.2 is working fine for me.
So your soldering might be the issue, try soldering headers instead.

KalyanSingh05, 2025-03-01:
@Lamnxzp when i am connecting the module to the m5 stickC plus 2, the led on the pn532 module glows that means module is started. correct?

Lamnxzp, 2025-03-01:
@KalyanSingh05 I don't know, but the LED does not light up for me.

KalyanSingh05, 2025-03-01:
@Lamnxzp ok then tomorrow i will redo all the connections

closed 2025-05-24 by bmorcelli