BruceDevices/firmware #763: want to know lilygo  T DECK  + CC1101/NRF Connection schematics

a5w1na5t60, 2025-02-01:
I haven't find t deck cc1101/nrf/rfid connection schematics. Please provide the schematics ASAP

bmorcelli, 2025-02-01:
It doesn't have

a5w1na5t60, 2025-02-01:
That means t deck doesn't support nrf and cc1101- I'm a t embed and t deck user so I have one more doubt can we took standalone connection and use cc1101

bmorcelli, 2025-02-01:
MAYBE it can be possible to use it with a SDCard Sniffer and the grove connector (43,44) as CS and GDO0 (CS and CE for Nrf24), but it probably wouldn't be possible with T-Deck Plus, because GPS uses UART on these pins

But you will need to uncomment these lines https://github.com/pr3y/Bruce/blob/main/boards%2Flilygo-t-deck.ini#L86

And https://github.com/pr3y/Bruce/blob/main/boards%2Flilygo-t-deck.ini#L95

And build the project for T-Deck

killerbe, 2025-02-05:
Module cc1101 does not work properly on frequency 433.92

Steps To Reproduce:
Caught a signal and saved it.
File - the signal did not play in the saved file.
It worked only when it was not saved.

Device used: M5StickPlus2
Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: 00 00 00 00 00 D5 12 28
TE: 249

Mrod31, 2025-03-09:
For NRF show here: https://www.thingiverse.com/thing:6963487

closed 2025-03-19 by bmorcelli