BruceDevices/firmware #766: Bad Ble,Media CMD'S,Blueetooth spam

Azuuu1, 2025-02-01:
**Is your feature request related to a problem? Please describe.**
Well mostly its just how bluetooth spam just barely works idk if theres any way to fix it or so but just saying

**Describe the solution you'd like**
Couple of things first off all with media cmds u should atleast list what device ur controlling or have a list of couple of them that u can pair to or something also if u connect to the spam thing if possible make it have an option to run scripts same as with media cmds make an option to run scripts

Azuuu1, 2025-02-01:
also maybe try to make bluetooth ios,android spam better as it barely works idk if its related to the m5stick or to the software maybe the nrf bluetooth module would help it but idk

Azuuu1, 2025-02-01:
and also please add more irs rather than just tb b gone like a flipper menu for the ir would be really nice

bmorcelli, 2025-02-01:
> with media cmds u should atleast list what device ur controlling or have a list of couple of them that u can pair to or something also if u connect to the spam thing

It uses standard media commands, common to almost all mobile devices

> if possible make it have an option to run scripts same as with media cmds make an option to run scripts

It crossed my mind once.. but I don't see anybody using it 

> also maybe try to make bluetooth ios,android spam better as it barely works idk if its related to the m5stick or to the software maybe the nrf bluetooth module would help it but idk

For the 1037392919374 time, IOS and Android have patched these type of attack, so the advertising won't happen every time.. it will be just annoying (somehow), bruce code is almost the same that runs on Marauder fyi

> and also please add more irs rather than just tb b gone like a flipper menu for the ir would be really nice with ac's projectors exc

Why would we pack more useless information into the flash memory (that is very limited) if you can get all these codes (.ir files) from as many repositories as you can, choose among them and use what you need? TV-B-GONE is a showcase, useful in 80% of the TV cases.. the other codes are better off the code.. spend sometime searching for your .ir files, or even reading, saving and sharing them.

closed 2025-02-01 by bmorcelli