BruceDevices/firmware #771: Always shows 100% charge on the battery (T-Embed CC1101)

TurboKoT1, 2025-02-02:
**Describe the bug:**
On my LilyGo T-Embed CC1101 i always see 100% charge on the battery, it doesn't change

**Steps To Reproduce:**
- Disconnect battery and connect it again.
- Device used: 
LilyGo T-Embed CC110
**Expected behavior:**
Normal display of battery charge percentage ¯\_(ツ)_/¯

**Actual behavior:**
 i always see 100% charge on the battery, it doesn't change

**Additional context / logs:**
Flashed my LilyGo T-EMBED cc1101 to Bruce firmware. At first - the percentage of charge was displayed correctly, but after disconnecting and reconnecting the battery - the charge always showed 100%

bmorcelli, 2025-02-02:
Battery value is updated when you move throughout the menu...

Battery is displayed as a calculus based on my device, so 100% is shown when above 1070mah.. while less than that, it shows percentage = charge/10.7

So leave it unplugged for some time and move across the main menu to update the value

closed 2025-02-03 by TurboKoT1

mileshkin, 2025-02-25:
Similar behavior. The battery indicator value never changes and always shows 100%, no matter how long the device is disconnected from charging and no matter how long you move through the menu.