BruceDevices/firmware #774: Wifi bruteforcer interpreter doesn't scan on the beta version

Gosheto1234, 2025-02-03:
**Describe the bug:**
I downloaded the beta version and when i click scan it doesn't scan

**Steps To Reproduce:**

- Device used: m5stickplus2

**Expected behavior:**

it should scan

**Actual behavior:**

it says no wifi networks found

**Additional context / logs:**

I downloaded/burned the newest beta today from https://bruce.computer/flasher and someone said that its fixed in the beta , i downloaded the file from the [Bruce](https://github.com/pr3y/Bruce/tree/main)/[sd_files](https://github.com/pr3y/Bruce/tree/main/sd_files)/[interpreter](https://github.com/pr3y/Bruce/tree/main/sd_files/interpreter)
/wifi_brute.js and when i run the .js on my stick it doesnt scan the wifi , is it only me or it doesnt scan ?

Lamnxzp, 2025-02-03:
https://github.com/pr3y/Bruce/blob/881817a3174cb96295036900c66181756b4ca854/sd_files/interpreter/wifi_brute.js#L38
Can you try to edit this line: `lenght` --> `length`

Gosheto1234, 2025-02-03:
it worked but it doesnt show networks , it shows only cancel

PR #775 (2025-02-03): Fix wifi_brute.js script
#### Proposed Changes ####
- All networks are now displayed correctly.
- Resolved multiple issues:
  - Removed `\r` characters from passwords list.
  - Fixed variable placement in the main loop that was unintentionally clearing stored values.
  
- JS Interpreter: Added 'waitKeyPress' parameter to `dialogMessage` and `dialogError`

Closes #774
closed 2025-02-03 by pr3y