BruceDevices/firmware #776: cc1101 is not working properly

killerbe, 2025-02-03:
**cc1101 does not work properly on frequency 433.92**

**Steps To Reproduce:**

- Device used: M5StickPlus2

Filetype: Bruce SubGhz File
Version 1
Frequency: 433919968
Preset: 1
Protocol: RcSwitch
Bit: 24
Key: 00 00 00 00 00 D5 12 28
TE: 249

killerbe, 2025-02-04:
> Did you save it on LittleFS or on SD Card?

Yeah, I save it on LittleFS.

AlexC-0806, 2025-02-06:
I used LittleFS and it has been working. With SD Card I didn't try yet, but I think it's more tricky.

atomov93, 2025-02-19:
any resolution here? I have the same problem

Huzzla101, 2025-02-19:
Did u try to read booth RAW and clean format to ,preset 1 maybe is raw? As
i use m5stickc plus2 myself and i can read raw easily but it have some
problems with clean formats but its a small issue tho, also put range to
only Fx 433.92 and raw format then it should manage most signals...

Huzzla101, 2025-02-19:
Just checked my lily go t embed and i had over 48 RAW signals captured in
under 40min.. I live near Centrum tho :p I will check the security of them
to see if my neighnours need any help off course..

atomov93, 2025-02-23:
I figured it out! Just download the beta. Its fixed there!

closed 2025-03-27 by IncursioHack