BruceDevices/firmware #779: Error E:5-Wrong Partition Scheme on 1.8.2 Bruce-LAUNCHER_CYD-2USB.bin using M5Stick-Launcher

Talismancer, 2025-02-04:
**Describe the bug:**
I just tried the 1.8.2 update Bruce-LAUNCHER_CYD-2USB.bin and got: "E:5 -Wrong Partition Scheme".
When I load the file of the same name from 1.8.1. No problems.
I then tried all the Launcher bins with same result.

**Steps To Reproduce:**
Install latest M5Stick launcher via web installer
Download release 1.8.2 Bruce-LAUNCHER_CYD-2USB.bin
Add bin to SD card with the requisite SD files
Install SD and power up CYD
Select the bin using the Launcher

- Device used: CYD 2432S028 2USB w/ USB-C

**Expected behavior:**
Bruce load screen appears with friendly shark

**Actual behavior:**
Error E:5 -Wrong Partition Scheme

BblackmilkK, 2025-02-04:
use beta version for launcher.

bmorcelli, 2025-02-04:
I just updated Launcher, removed OTA for non M5 devices..

Install the Last Release (or Beta Release, they are the same atm) usinf the webflasher

closed 2025-02-04 by bmorcelli