BruceDevices/firmware #785: SD CARD Turn-off Mode Sleep

akeendev, 2025-02-05:
I would like to report an issue found in version 1.8.2 related to the SD card initialization process. When the LilyGO CC1101 device is powered off using the 'Turn-off' menu in Bruce and then powered on again, the SD card is not initialized correctly. This happens because, when using the 'Turn-off' option, the device enters sleep mode. Upon waking up and powering on again, the device does not reactivate the SD card reading. The SD card initialization only occurs again if the device is powered off by removing the battery and turning it on again, or by removing and reinserting the SD card.

bmorcelli, 2025-02-17:
I added Full turn off in my brench, can you test it, please?

available on: https://bmorcelli.github.io/Bruce/flasher > BetaRelease > T-Embed CC1101

to fully turn off, go Config > Turn Off

the shortcut (upperbtn and encoder roll still does Deepsleep

akeendev, 2025-02-18:
The Turn-off works, but when using deepSleep, the SD card is not mounted when powering back on.

akeendev, 2025-02-27:
After using an 8GB SD card, it works, returning from sleep mode with the SD mounted. I believe SD cards larger than 8GB don't have enough time to remount after waking up from deep sleep.

closed 2025-05-24 by bmorcelli