BruceDevices/firmware #789: LilyGo T-Display S3

zynwire, 2025-02-06:
Hey, I recently installed BruceFW on my LilyGo T-Display S3 but how do I select stuff? Can someone send a little cheat sheet of some sorts for what to do for what things since working with 2 buttons only is hard lol.

ezicvtumane, 2025-02-06:
You need to solder an additional button to pin 16. She will serve Select button in PIN to 16

zynwire, 2025-02-06:
Can i just use a jumper or smth?

ezicvtumane, 2025-02-06:
To check, I first simply short-circuited 16 pins to ground using a wire

zynwire, 2025-02-07:
Epic

djaysan, 2025-02-23:
t-display s3 here, i understand that we need to short pin 16 & ground to select an option. 
 How de we wire an sd card reader? 
 
 I did that on t-display 1.14 following [this guide](https://github.com/ATOMNFT/Marauder-T-display-TTGO?tab=readme-ov-file#esp32-marauder-info) but i suppose Bruce is different and the pi numbers as well

epiHATR, 2025-02-24:
is this for t-display s3 or just t-display @djaysan  ?

djaysan, 2025-02-24:
@epiHATR i have both :
1. I flashed **Bruce** on my t-display **S3** - i'm trying to get **Bruce** running with a select button (which works on pin 16 > ground) as well as wiring an **SD reader** and **GPS** module (Rx, Tx). but no clue where to wire them. 

2. On the other hand I have an t-display **ttgo 1.14** (the smaller brother) with **Marauder** installed, I don't need an additional button since the menu just goes forward with one button and the other button is to 'select' an option. I was able to wire an **SD card** and it works fine. Next will be the **GPS** as well for some _wardriving_.

Karxthi, 2025-04-21:
@djaysan 
could it be possible for you to share the schematics of the 1.14 one ?

djaysan, 2025-04-27:
> [@djaysan](https://github.com/djaysan) could it be possible for you to share the schematics of the 1.14 one ?

I don't have bruce on my 1.14 one - only on the t-display S3. so far, i got the select button, gps wired up and i will try the sd card.