BruceDevices/firmware #792: Power and battery issues on JC2432W328C

RChadwick7, 2025-02-06:
I just installed Bruce on a JC2432W328C, and mostly works great! I'm happy to see the capacitive screen drivers working good.

However, when connecting a battery, the button turns the device on, but not off. I'm guessing this is a hardware limitation. Could there be a menu choice to shut down the unit through software? Or look for the power button and shutdown then?

Also, the battery is always at 0%, not sure if there is hardware to read the battery but I'm sure it wouldn't be too hard to add.

bmorcelli, 2025-02-15:
it doesn't have this measure, so it'll always display 0%

dankcincy, 2025-03-18:
Did you ever find a solution for the button issue?