BruceDevices/firmware #794: RFID Lilygo cc1101 Clone UID/Write data not working

akeendev, 2025-02-07:
Bruce versão 1.8.2: os módulos RFID "Clone UID", "Write Data" e "Erase Tag" não estão funcionando corretamente.

Ao utilizar uma tag de 13,56 MHz com o bloco 0 regravável, não é possível realizar as operações de clonagem de UID ou gravação de dados. O sistema retorna as mensagens de erro:

"Error writing UID to tag"
"Error writing data to tag"

Além disso, o módulo "Erase Tag" informa sucesso na operação, mas a tag não é apagada corretamente.

akeendev, 2025-02-10:
Regarding the error, if I do a read tag and then use the write tag and bring the same tag close, it performs the write on the tag. However, if I use another tag, it gives an error when writing.

CharlesSeng, 2025-03-01:
Yeah, me too. I have the same issues. One I always get is ‘failed to read data blocks’ when reading any tags. Please rectify the issue if possible.

Cloveeee, 2025-03-08:
Hey mines not reading tags at all :(

CharlesSeng, 2025-04-07:
So how? It seems like the admin has no clues about how to fix this bug. Unfortunate…

mvellinger, 2025-04-07:
Well, I don't know about what's on your mifare card of course, in my case the one I'm reading actually only has uid data on it in block 0, nothing in the subsequent blocks, and I suspect the error message is actually wrong (i.e. it correctly read the card, there just wasn't anything _to_ read in the subsequent blocks)

As an example, this is (part of) the dump of the test card I'm reading in the attachments 

[image]

closed 2025-05-24 by bmorcelli