BruceDevices/firmware #797: Tembed cc1101 problems

nikogutierrez, 2025-02-08:
I have a t embed cc1101 and the rf module is recognized and can receive rf signals but can not transmit them. For example my saved rf signals are not actually transmitting and the jammer does not work. I have checked that the connections are good on the inside and the antenna is connected.

varenichek22, 2025-02-09:
Can you show any logs via Serial? Also the cc1101 antenna is not the brightest star, rolling codes can be a problem.

atomov93, 2025-02-19:
same problem here. The code is not a rolling code. I copied it on an universal remote. When I replay it in RF its fine. Not from a file

Huzzla101, 2025-02-19:
I just checked RF on m5stick c plus2 and lilygo Tembed and i did manage to
send back and forth RAW signals easily and booth devices captured and did
send so that should not be an issue... But we should also when you master
coders have time check out the "clear" format not raw as i have not
received that yet and also when i send from my portapack h4m a 433.92 .sub
when lilygo Tembed cc1101 is on Fx 433.92 raw it picks signal up but when i
take the 370\460 +- range or even the "ALL RANGES" settings it do not pick
up the same 433.92 .sub signals yet... But i know you guys will check it
out when u have the time !!! =)  Peace love and harmony 🦈🦈🦈

Huzzla101, 2025-02-19:
I need to add that also the RF JAMMER booth of them , works fine .. They
even put out my Portapacky and the NRF24 JAMMER is smooth as a mother
trucker.. NRF24 to your door,no bluetooth no more.... :p

atomov93, 2025-02-23:
I figured it out! Just download the beta. Its fixed there!

closed 2025-03-27 by IncursioHack