BruceDevices/firmware #798: cc1101 catches signals from nowhere

Metinsz, 2025-02-08:
**Describe the bug:**

so when i go to the copy section it starts spamming signals over and over again i tried jamming to but no hope i do not think its about the module because it came today and i didnt do anything to it also
idk if its normal but spectrum reacts alot when i touch the antenna pls tell me if its becuase of the cables

Azuuu1, 2025-02-09:
u have to stop reading raw signals then it will stop i had the same issue.

GthiN89, 2025-02-09:
if it have constant false reception of subghz signals, i came across this problem today too... I already reported it, with provided solution...
https://github.com/pr3y/Bruce/issues/802

Metinsz, 2025-02-10:
> I tried to ground GDO2 here, and it didn't make any difference.. I usually get these random signals when I have false (or poor) connections..
> 
> I always suggest people to use short wires ( max 10cm)... bigger wires work as small antennas driving

> I tried to ground GDO2 here, and it didn't make any difference.. I usually get these random signals when I have false (or poor) connections..
> 
> I always suggest people to use short wires ( max 10cm)... bigger wires work as small antennas driving

ill try making a board for it hope it gets fixed

Metinsz, 2025-02-11:
> I have this issue since i started using od DC DC step up converter, generally, it helps to ground everything you can, and put decoupling capacitator everyvhere you can. When this issue arrive, i just try grounding and adding those decoupling caps, until it solves....

ill try making better connections first because it might be the cables or the module idk but i will definitely try adding more grounding if it dosent work

closed 2025-02-17 by bmorcelli

GthiN89, 2025-02-22:
With 150 ohm external pull up resistor on CC1101 GPIO pin (RX/TX), decoupling capacitator, being grounded on small "ground plate", so far, works perfect every build.

bmorcelli, 2025-02-22:
Can you send the schematics?

GthiN89, 2025-02-22:
schematics looks like this, i will sen photos from phone in a minute...

[image]

GthiN89, 2025-02-22:
LOL i uploaded photos here, sorry...
https://github.com/pr3y/Bruce/issues/802#issuecomment-2676235906