BruceDevices/firmware #799: Ir issues

Yeasyease, 2025-02-08:
Custom ir barely works after updating from 1.8.1 -> 1.8.2

Enter custom ir and try to control tv of choice.
- Device used:  Ir transmitter module and m5stickcplus1.1

I expeceted it to on the first attempt to turn off.

I need to send the signal multiple times before the tv actually receives it.

I think the issue could be that Ir signals in general seem to be sent faster than previous version, perhaps make a setting in ir config where you can change the speed it sends the signals to counter this issue.

Lamnxzp, 2025-02-09:
Hey @Yeasyease,
Are you sure you were on Bruce 1.8.1?

Some things changed here: https://github.com/pr3y/Bruce/commit/bbdb636cb510dbc65907f0b68663e60221f4315c
Now, unnecessary repeats have been removed (11 signals sent → 1 signal sent).

However, this change also applies to Bruce 1.8.1, so maybe you were actually on Bruce 1.8?

Yeasyease, 2025-02-09:
It says the protocol is Samsung32
The specific file has worked before updating

Yeasyease, 2025-02-09:
And yes before I was using 1.8

Lamnxzp, 2025-02-09:
@Yeasyease  
It’s probably because you’re too far from your TV or the IR transmitter isn’t strong enough.  

I could add a setting that lets you adjust the number of repeats sent (so it will works the same as before).

Yeasyease, 2025-02-09:
Please do so in a further update

Yeasyease, 2025-02-09:
Okay. It's inconsistent even with the integrated ir transmitter from the m5stickcplus1.1

Lamnxzp, 2025-02-09:
It's working fine on my Samsung TV...

Yeasyease, 2025-02-09:
At least make it a option for compatibility sake.
How long will it take to release the fix?

Lamnxzp, 2025-02-17:
Hey @Yeasyease ,
I made the changes in #821
You can try on https://lamnxzp.github.io/Bruce/flasher -> Beta Release -> M5Stack -> StickCPlus 1.1 or StickCPlus2
I made a new setting in IR -> Config: "IR Tx Repeats", select 10 and it should work the same as before.

PR #821 (2025-02-19): Add IR Tx Repeats support, fix IR on Serial Cmds
#### Proposed Changes ####
- Serial Commands:
  - Added support for all decoded protocols.
  - Added support for RAW type.
- IR:
  - Added "Ir Tx Repeats" setting to specify the number of additional repeats (1 initial + X repeats), when sending a command from Custom IR.
  
#### Linked Issues ####
Resolves #799
closed 2025-02-18 by Yeasyease