BruceDevices/firmware #800: Bruce doesn't send all the IR-Codes

TurboKoT1, 2025-02-09:
**Describe the bug:**
Bruce doesn't send all the IR-Codes

**Steps To Reproduce:**
Download https://github.com/UberGuidoZ/Flipper/blob/main/Infrared/tv.ir, send POWER cmd -  and Bruce will only send the first power code - when he should be sending each one (94 codes)
- Device used: 
LilyGo T-Embed CC1101
**Expected behavior:**
Sending all IR Codes

**Actual behavior:**

Sending only first IR Code
**Additional context / logs:**
-

Lamnxzp, 2025-02-09:
Hey @TurboKoT1,
'Select cmd' is meant for sending only one command at a time.

You can either use 'Spam All,' though this will also send 'VOL+,' ... 
Or edit the IR file to include only the 'POWER' commands.

Lamnxzp, 2025-02-09:
Your file (https://github.com/UberGuidoZ/Flipper/blob/main/Infrared/tv.ir) is meant for use with the 'Universal Remote' feature on the Flipper Zero.

closed 2025-02-09 by TurboKoT1