BruceDevices/firmware #801: Universal Remote (as in flipper zero)

TurboKoT1, 2025-02-09:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when Bruce can't use universal remote files like Flipper Zero (https://github.com/UberGuidoZ/Flipper/blob/main/Infrared/tv.ir)

**Describe the solution you'd like**
Universal Remote function, which will work something like this:
User selects a button (POWER, VOL+, VOL-) etc -> Bruce starts trying to send all ir codes from the file

**Describe alternatives you've considered**
-

**Additional context**
-

TurboKoT1, 2025-02-09:
That's a bit of a misnomer.

I mean that the user selects a function (POWER, VOL+, VOL-) and Bruce starts searching all IR Codes of the same function, not all of them.

bmorcelli, 2025-02-10:
By what you are talking, you might be accessing the .ir file from the "Files" menu.. and this is the behavior when shooting from there..

Try going: **IR > Custom IR > Sd Card (or LittleFS) > (Find your file) > Choose cmd**

There you can choose up to 100 commands from the file.. but I suggest to use smaller files to avoid crashes

bmorcelli, 2025-02-10:
In this case it is easier for him to set up a JavaScript calling all the commands and send it from there..

There's no reason for us to hardcode all these commands into the firmware (TV-B-GONE is the  exception)

Lamnxzp, 2025-02-10:
@bmorcelli You're right. But I think the user wants a way to load their IR file and showing "Power", "Ch+", "Ch-", "Vol+", "Vol-", "Mute", that would send all matching commands from the IR file.

However, it might be confusing for some people to add a 'Universal Remote' that requires the user to manually import the IR file.

Lamnxzp, 2025-02-10:
Alternatively, we could create a 'Universal Remote' feature that looks for IR files in an 'assets/ir' folder.
For example, if a user has 'tv.ir' in the folder, it would show 'TV' as an option and then display an interface similar to the Flipper Zero Universal Remote.

[image]

IncursioHack, 2025-03-27:
@pimentelleo PSC.

pimentelleo, 2025-03-27:
In theory, he wants a function that recursive spam all codes that have the same name on the desired button. That is required for an "universal" remote since the universal spams all codes for a certain button, then should work with any device. Is that how a normal "universal remote" works. 
I will look into it but for now I am working into the web remote 
This should be simple since if we wants to trigger a button, and it have more entries on the IR file, then spam all that have the same. This can be the normal behaviour for almost any ir file.

CloudWaddie, 2025-04-29:
Please add this