BruceDevices/firmware #802: Bad CYD+CC1101 Schematics

GthiN89, 2025-02-09:
**Describe the bug:**

Mistake in wiring can cause floating pins, in my case CC1101 GPIO1
**Steps To Reproduce:**

- Device used: Just start implement oldschool radioamateur soldering - circuit building techniques, im not able to give you exact steps, this just happens... You can like have it working in one build, and have this "bug" in another one...

**Expected behavior:**

Do not get false SubGhz signal reception every second.
**Actual behavior:**

Constant SubGhz false signal reception
**Additional context / logs:**

Solution is pretty easy... You just fallow basic digital circuit rules (im not some expert, i started MCUs like 8 months ago, i know like just basic rules like this one), so you do connect all unused gpios to the (common) ground. Just like this....
Problem solved.... Pleasse update your wiring diagrams ;-)

[image]

GthiN89, 2025-02-10:
you can try ground those circles coated with tin, and outher connector of anthena. do you have something what can introduce lot of noise? Like step up converter?

GthiN89, 2025-02-22:
With 150 ohm external pull up resistor on CC1101 GPIO pin (RX/TX), decoupling capacitator, being grounded on small "ground plate", so far, works perfect every build.

GthiN89, 2025-02-22:
[image]

[image]

[image]

[image]

[image]

Photos

GthiN89, 2025-02-22:
ground plate is just piece of non itched PCB covered in tin...

Huzzla101, 2025-02-22:
Haha  that is a beast in disguise tho... Dayyyum i need to step up my
game... Keep it coming that was smooth... =)

closed 2025-03-28 by bmorcelli