BruceDevices/firmware #808: CC1101 won't work with M5StickCPlus2

kilumao, 2025-02-12:
I'm using Bruce Firmware and connected CC1101 onto it and I think the CC1101 can be found because it lets me go into the spectrum and doesn't give me any error messages but does not work. But also I used the same CC1101 on my Flipper Zero and it worked perfectly fine. So I think it is an m5stickcplus2 issue idk? Someone, please help me!!! Also,o it used to work but now it does not work on my m5stickcplus?!!! helpppp

SpokHacker, 2025-02-13:
Have you tested this on other firmware?

kilumao, 2025-02-13:
Not really I tested it on CatHack but I think they use a different pinout. I think its a m5stick issue???

SpokHacker, 2025-02-15:
It's working fine for me

kilumao, 2025-02-15:
not for me can someone help