BruceDevices/firmware #816: Built in text editor

iacobucci, 2025-02-14:
**Is your feature request related to a problem? Please describe.**
Within the bruce firmware its possible to explore, rename and move files, but its not possible to write text in them.

**Describe the solution you'd like**
It would be cool to have a built in text editor, maybe with syntax highlighting for javascript applications, for the cardputer. Maybe this wouldnt be viable for other devices, though...

**Describe alternatives you've considered**
There is this same functionality in MicroHydra for cardputer ([echo-lalia/MicroHydra](https://github.com/echo-lalia/MicroHydra))

**Additional context**
![](https://private-user-images.githubusercontent.com/108598670/323036975-15b78e4b-64fc-433a-86d3-979362abd9ab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk1NDI3NjcsIm5iZiI6MTczOTU0MjQ2NywicGF0aCI6Ii8xMDg1OTg2NzAvMzIzMDM2OTc1LTE1Yjc4ZTRiLTY0ZmMtNDMzYS04NmQzLTk3OTM2MmFiZDlhYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxNFQxNDE0MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYjE5NjU1ZTU1NjE1MmQ4MjljYjdkYWFiZWU4NDA2YzFmYWY2MmM0Y2FlNDI4ZjNjYTk2N2U4MzQ4MDhmNDU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.o6_h6xaj7d2y87Tb7Ar5IhsUWHYh_LUst2Zto9St1A0)
Look at how the first 3 and the last 3 lines get "squashed": the font is smaller for those 6 lines, so that the user can still see them but is more focused on the central lines. Its the best you can do with the not so generous screen real estate of the Cardputer.

closed 2025-05-24 by bmorcelli

iacobucci, 2025-05-24:
I understand that on devices like the Lilygo T embed it wouldn't make much sense, but that's a real bummer for devices like the Cardputer or the Lilygo T Deck, that could make a great use of their full qwerty keyboard with a simple text editor!

bmorcelli, 2025-05-24:
For those I made the Launcher