BruceDevices/firmware #827: Touchscreen Inconsistency

Sflofler, 2025-02-18:
**Describe the bug:**

I'm having some inconsistency with the touch screen. I'm currently building "my own" device, which is an ESP32-S3 with a `TFT` screen with  `XPT2046` touch controller, the rest of the modules are pretty much the ones mentioned in the docs.

I also have a CYD device on me and for the sake of curiosity I installed bruce on it just through the web flasher to see If I get the same behaviour, which I did!

**Steps To Reproduce:**

- Device used:
    - Own ESP32-S3 with TFT display(ILI9341/XPT2046)
    -  CYD 2432S028 (Dual USB)

So I found a way to easily reproduce it and by the looks of it it is always something that is looping somewhere in the code. The ones that I used to test it is the `Brucegotchi` and the `Read Tag` in the `RFID` menu. These menus are a struggle to get out of it, I have to press the screen multiple times or even hold it until it works, the problem in the `RFID` menu is that while pressing the screen a new option screen shows up but the input is not consumed so I accidentaly enter the `Clone` menu, It takes some tries until I can get access to what I want to in this menu.

I searched for references of `EscPress`, `SelPress` and other and I didn't found these variables beign used in these menus, I'm not entirely sure how input handling is implemented in these screens but let's take into consideration the `Clock` screen, It uses these variables and it works fine for me.

SPI is shared betwheen display and touch controller with separate CS pins.

These are my SPI settings(at the time of writing this):
```
#define SPI_FREQUENCY        10000000
#define SPI_READ_FREQUENCY   2500000
#define SPI_TOUCH_FREQUENCY  2000000
```
I did mess with it a lot so this is the current state but I also tried with the default values for CYD in the `.ini` files.

**Expected behavior:**

These menus shouls be responsive in only one touch.

**Actual behavior:**

Menus doesn't respond/ignore an certain amount of inputs, in this case in a resistive touch screen.

**Additional context / logs:**

It doesn't log anything actually but I did put a log when I touch the screen just to check if it is working and yes, It always logs the X and Y positions where it was touched.

Sflofler, 2025-02-18:
Even though the input log is fired normally? I guess this also applies for RFID menus, NRF24 Spectrum and other CPU hungry stuff?

Sflofler, 2025-02-19:
Defining `USE_TFT_eSPI_TOUCH` seems to fix the issue for my device.

IncursioHack, 2025-03-21:
@Sflofler Could you bring more details about this? If yes and the fix is functional (without breaking the other devices), please submit a Pull Request. If you have already done so, let me know so I can close the issue.