BruceDevices/firmware #828: Changing icons on Bruce

ACI077, 2025-02-18:
Guys, please help. You can make a detailed and step-by-step tutorial on changing icons in Bruce. (What is written in the wiki is not very detailed)

bmorcelli, 2025-02-18:
Changing Icons? 

What icons are you trying to change?

ACI077, 2025-02-18:
I'm trying to change the main icons and make them look like on a flipper. If I find a picture, I'll try to send it how it should look.

ACI077, 2025-02-18:
@bmorcelli Can you add a function to change the background?

bmorcelli, 2025-02-18:
Open the WebIU, and click in the pencil icon to edit bruce.conf

[image]

Edit the "bgcolor" from "0" to the color you want.. it must be from "0" to "FFFF", it is RGB565 (also known as 16bit RGB) format

[image]

Click Save, at the bottom of the screen, and restart your device

ACI077, 2025-02-18:
Thanks. But how can i change the main icons

ACI077, 2025-02-18:
No. I just changed bg color in #ff8200 and sec color, pri color in 0

ACI077, 2025-02-18:
@bmorcelli How can i change the icons?

bmorcelli, 2025-02-18:
> No. I just changed bg color in "ff8200" and sec color, pri color in "0". Try it

This format is not supported.. it must be 16bit format, use "fc00" instead

> How can i change the icons?

You don't change, they are drawn by code..

ACI077, 2025-02-18:
@bmorcelli I can make the icon code. Just tell me how to change the code and then download the firmware.

closed 2025-02-21 by rennancockles