BruceDevices/firmware #829: Changing icons on Bruce 2.0

ACI077, 2025-02-18:
Guys, please help. You can make a detailed and step-by-step tutorial on changing icons in Bruce. (What is written in the wiki is not very detailed)

I'm trying to change the main icons and make them look like on a flipper. If I find a picture, I'll try to send it how it should look.

I can make the icon code. Just tell me how to change the code and then download the firmware.

Huzzla101, 2025-02-18:
Wow that is on fire.. Pure magic ! =) I believe that 2025 is going to be
The Year Of The Bruce.. We taking over the world together with Brother
Flipper and all them goodie devices ...  People need to understand.... That
Bruce is tha Sharky Man...

bmorcelli, 2025-02-18:
We don't want Bruce to have the face of Flipper.. we want it to have Bruce's identity.. 

I'm sorry buddy, but no dev will help you with that..

ACI077, 2025-02-18:
Anyone can help me?

ACI077, 2025-02-18:
@bmorcelli Ok i want add a new function. How can i do it?

bmorcelli, 2025-02-18:
Learn C++, fork the repo, add your feature, test  and open a Pull Request.

ACI077, 2025-02-18:
And how can i fix that

[image]

closed 2025-02-21 by rennancockles