BruceDevices/firmware #835: esp32 CYD2USB/cc1101 device reads raw and jammer works but replay doesnt work on static code

tegtheevi, 2025-02-19:
hello i need help with this weird case so i hooked up a cc1101 to CYD2USB and it works perfectly it captures and jammes but when i try to replay a 433mhz signal nothing happens my garage door opener uses static code btw and i find it kinda weird not working anyone has the same problem 
the reciver is nice a01 garage door opener with static code 
present : 0
protocol : RAW
and the raw data is just numbers
thank you for your time

Huzzla101, 2025-02-20:
Most likely ur garage uses rolling codes.. I suggest u search on youtube
for FOB garage rolling codes and even add keys if u want to get some proper
explanation and smooth tips--

Huzzla101, 2025-02-20:
The static is clean format every time ? As most even static have a small
part before and / or after  main signal that changes... FOB can be a pain..

tegtheevi, 2025-02-20:
@Huzzla101 i researched the reciver and it a static code and the .sub files captured fromn the remote all match the same signal 
i tried to remove my car battery and capture the next unlock signal and put the battery back on and tested and it didnt work

Huzzla101, 2025-02-20:
There is one good video but i dont recall what link but he actually takes a
static garage fob and show us thru a raw deep RF spectrum analyser(it
records all data sendt out not just the specific key that a normal raw
reader will do) that its a small part before the same static key that
changes (same as dynamic keys , rolling keys but more hidden) ... As he had
same issue as u first then he found at that he needed to copy the small
part before the actuall key that raw format picks up, as that DOES change
every time and since its the same key everytime he could take the first
part and as long as that part was not read by recvr then he could use that
part together with keysignal, merge them as one and open it... I assume you
stand in front of the same great problem... Its a challenge we need to face
:p

closed 2025-02-20 by tegtheevi