BruceDevices/firmware #836: Flasher problem

ACI077, 2025-02-20:
Please @bmorcelli add the ability to change Bruce's versions to the site. Because I don't like this version of Bruce. Either add the function to change the firmware version, or add the function to change Bruce for yourself. I don't like that the Web UI feature is located in the files section. Also remove the label Booting and add a function to change the icons. Not everyone likes the standard icons.

rennancockles, 2025-02-20:
Why do you have to be so annoying? Stop tagging contributors to ask for changes that you like/dislike, you're not the PO of the project. It's an open source project by the way. If you don't like something and want it different, just create a fork of it and change whatever you like. Also, stop asking to change the icons, it's not supposed to be a flipper clone. If you want a flipper just buy one. Last, but not least, stop demanding stuff and start contributing.

ACI077, 2025-02-20:
First of all, I didn't ask you to change the icons, I just asked the person in detail how to do it. Secondly, I'm not asking for anything in return, I just want to offer you some ideas that could help in the development of the project. Third, I just asked you to please people (choose the Bruce version yourself). Last but not least, almost the entire project is a repeat of flipper.(the functions are the same, Bruce's symbols are a shark, and flipper's symbol is a dolphin) You just made an open source project using a stick, a cardputer, and other boards.

rennancockles, 2025-02-20:
You don't want to offer ideas, you want us to do stuff the way you want. As you said "change stuff because I don't like Bruce's last version". I don't know if you know how git works, but it's a version control system, all Bruce versions are there for you to download and flash whatever you want. Also you have all versions on M5Burner. And it's not because we have some functionalities just like flipper that it's supposed to be a clone. If so, all televisions are clones as they offer the same features. Same goes for refrigerators, ovens, ...

ACI077, 2025-02-20:
First of all, companies offer different refrigerators or televisions because it's a competition. You are just enthusiasts who wanted to create a system similar in functionality, and chose flipper. I don't mean Bruce separately. The entire firmware looks like a flipper. Secondly, the versions are not available in the m5 burner. Only the current version is displayed there. Thirdly, I asked to add this feature because I don't like the current version, another user may have another reason to install the previous version. In conclusion, you're just afraid to share details with us about how to change icons or add features because you don't want people to be able to make the firmware better, more reliable, and more powerful. I asked you 4 times to write a detailed tutorial on how to do this, but you ignore people and turn away from the answer.

closed 2025-02-20 by bmorcelli