BruceDevices/firmware #838: Add files to LittleFS

12-AD, 2025-02-20:
When compiling the source, where can you add files to littleFS so they are there every time you compile? Wanna use this to make my own version with some files and custom config that stays whenever I recompile. (NO SD CARD)

bmorcelli, 2025-02-20:
Create a folder called "data" at the root of the project, and add the files you want in the little fs there.

I believe that it solves your problem, tell me if it works

lobolinux, 2025-02-20:
LittleFS is very good. I'm having problems compiling evilcrow V2, and I was instructed to install LittleFS. If you could give me a step-by-step guide for this, I would be grateful.

Case: 
I'm having problems installing Evil-Crow-V2.

I installed it from https://github.com/joelsernamoreno/EvilCrowRF-V2, but when compiling it shows an error:

Compilation error: SPIFFSEditor.h: no such file or directory.

However, I didn't move the file or directory.

Sorry for Offtopic, but my case can help others too.

bmorcelli, 2025-02-21:
> Compilation error: SPIFFSEditor.h: no such file or directory

Evilcrow firmware depends on older library version of ESPAsyncWebserver... You need to install version 1.3.1 of this library and use Esp32-Arduino framework below 2.14 to work

lobolinux, 2025-02-21:
Amigo @bmorcelli 
I searched in git and other places and didn't find both:
- ESPAsyncWebserver - version 1.3.1
- Esp32-Arduino framework below 2.14

Could you help me?
tks

bmorcelli, 2025-02-21:
https://github.com/esphome/ESPAsyncWebServer/archive/refs/tags/v1.3.1.zip

If you are using Arduino IDE 1.8, it is not a problem .

If using Arduino 2.x, bo to the "boards" tab, where you installed the ESP32 boards, and change the version to 2.14 or 2.17

lobolinux, 2025-02-22:
Amigo @bmorcelli 

/Users/XXXXXX/Documents/Arduino/libraries/EvilCrowRF-V2-main/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino:168:1: error: 'esp_task_wdt_config_t' does not name a type; did you mean 'esp_task_wdt_init'?
 esp_task_wdt_config_t wdt_config = {
 ^~~~~~~~~~~~~~~~~~~~~
 esp_task_wdt_init
/Users/XXXXXX/Documents/Arduino/libraries/EvilCrowRF-V2-main/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino: In function 'void force_reset()':
/Users/XXXXXX/Documents/Arduino/libraries/EvilCrowRF-V2-main/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino:610:22: error: 'wdt_config' was not declared in this scope
   esp_task_wdt_init(&wdt_config);
                      ^~~~~~~~~~
/Users/XXXXXX/Documents/Arduino/libraries/EvilCrowRF-V2-main/firmware/v1.3.3/EvilCrow-RFv2/EvilCrow-RFv2.ino:610:22: note: suggested alternative: 'tmp_config2'
   esp_task_wdt_init(&wdt_config);
                      ^~~~~~~~~~
                      tmp_config2
Multiple libraries were found for "SD.h"
  Used: /Users/XXXXXX/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/SD
  Not used: /Users/XXXXXX/Library/Arduino15/libraries/SD
exit status 1

Compilation error: 'esp_task_wdt_config_t' does not name a type; did you mean 'esp_task_wdt_init'?

12-AD, 2025-02-22:
My problem was fixed, I’ll close the issue in 24h, so if you guys want to keep talking I recommend moving to the discord server or something.

12-AD, 2025-02-22:
Actually will not be closing it as the solution provided did not work

12-AD, 2025-02-22:
Looked though the code for about an hour 😓 the solution if I’m not mistaken is to make a folder called “LittleFS” and in it make one called “data”

12-AD, 2025-02-22:
FOUND THE ISSUE!!

Create a "data/"Folder in root directory

Place file inside data/.

double check that this line below is somewhere in your platformio.ini file
``board_build.filesystem = littlefs``

then run this: 

``pio run --target uploadfs``

(if im not mistaken this adds the file for all devices. if you want to add it to only one you can do something like this)

``pio run --target uploadfs -e M5StickC_Plus2``

closed 2025-02-22 by 12-AD