BruceDevices/firmware #840: Hi dude,

ThundeeMax75, 2025-02-22:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

Sory, I can't use discord in Turkey, with that reason I need to write from here.

I m just new around. I suppose to use m5stack hardware for my job (qa inspector)  I discovered your firmware and fall in love it. And after that I ordered a stickc plus 2 and a Cardputer with rfid and qrcode units. Also I connected GY-GPS6MV2 module with grove too, works perfect. I need a future and any firmware got this. Could you add a section to firmware for qr-code reader. Maybe less people need this future but any firmware got this feature. Thanks in advance.

ThundeeMax75, 2025-02-22:
My mail address, m.akif.a@gmail.com

bmorcelli, 2025-03-18:
You want to read the qrcodes displayed on bruce screen, ryt?

If is that what you need, it is available on Bruce (since many versions)

It is in: Others>Qr codes

You can add new codes editing bruce.conf file

If it was helpful and solved you problem, close the issue 😜

ThundeeMax75, 2025-03-18:
Hi all,

I think I could not explain my problem, I would like to use a hardware (m5
stack qr-code unit)  with bruce firmware. I'm not professiona code
developer but, I developed simple code (probably need some optimize, I
added working video), adding code to this mail. also adding video, photos
and some screenshots of code. If it's possible please add this code to
firmware under extra section or somewhere, so people can capture qr code
datas with bruce firmware with official M5 qr code hardware. I try to send
files with mail but get error. So added all to wetransfer.

https://we.tl/t-5EhMaOsD86
[image: error.jpg]

closed 2025-05-26 by bmorcelli