BruceDevices/firmware #842: Storing wifi passwords in the configuration in an unprotected form

cryptexctl, 2025-02-23:
**Is your feature request related to a problem? Please describe.**
When I decided to look at bruce.conf, I was shocked to see that the wifi password is stored in memory in clear form, and anyone can get it from there if they want to.

**Describe the solution you'd like**
I suggest encrypting it somehow, with an additional master password for example, or at least base64.

Lamnxzp, 2025-02-23:
> ... and anyone can get it from there if they want to.

No, not "anyone". Accessing `bruce.conf` requires connecting to the AP (if AP Mode is used) (which requires a password), then logging into the WebUI with valid credentials, so it’s already well secured.
*You can easily change all these credentials in the `bruce.conf` file.*

> I suggest encrypting it somehow, with an additional master password for example, or at least base64.

Base64 is just an encoding method, not encryption, [it's easily reversible](https://www.base64decode.org/).
Adding a "master password" would require users to enter it each time Bruce needs to access these WiFi credentials (or once at startup and kept in the memory). This adds complexity without significantly improving security.

cryptexctl, 2025-02-23:
Is it theoretically possible to pull information from the board's memory?
If it can, it is still a problem
If no - then we can consider that there is no sense in adding additional wifi password protection in the configuration.

bmorcelli, 2025-02-23:
In theory, it could use the HWID (which is the physical MAC Addr of the ESP32), that is "unique", to use as primary key.. 

But you wouldn't be able to share the SDCard with other Bruce devices (which would be a problem for me, lol), because the key would be different...

bmorcelli, 2025-02-23:
> Is it theoretically possible to extract files from the memory located on the board/memory card(?)?

You can dump the memory with esptool and analyse the binary... You can retrieve some  info you you know where to look... So yes, it is possible... And the SDCard is just open in a computer with notepad, or vim or whatever text editor..

cryptexctl, 2025-02-23:
> wouldn't be able to share the SDCard with other Bruce devices (which would be a problem for me, lol)

I think it would be possible to make some sort of switch in the configuration with the option to disable encryption of wifi network passwords.

Lamnxzp, 2025-02-23:
@cryptexctl Someone could just go to "Files -> LittleFS" and see the bruce.conf content.
So yeah.. but why would you give your device to someone you don't trust?

cryptexctl, 2025-02-23:
[image]

Also after a bit of research, I realized that the full configuration is also output to the UART where the password can also be retrieved without problems

cryptexctl, 2025-02-23:
> ... someone you don't trust?

It's just not a good thing when _private_ information in the form of a password that should not be known to anyone but the device owner is shared in the open, and virtually anyone with the desire to do so can view that _protected_ information from there.

rennancockles, 2025-02-23:
That's simple. You can either stop connecting to WiFi, that way you won't get your password stored into FS, or create a PR with a solution you think is good enough.