BruceDevices/firmware #846: no display/boot on t embed cc 1101

srgggaa, 2025-02-25:
i Am trying to install bruce i tried a lot of things but i cant get it to boot/ schow a screen but other firwears that are dogschit like capibara zero work fine so i need help installing bruce and it worcking

(i am dislexic)

bmorcelli, 2025-02-25:
* Take the back cover
* Connect to USB with a Data cable
* Press and hold the encoder button (center btn of the wheel)
* Poke (press and release) the RST button inside the t-Embed CC1101
[image]

* Release the encoder btn
* Go the https://bruce.computer/flasher > BetaRelease > Lilygo > t-Embed CC1101 > Connect > Install ....
* Wait for it to erase and install
* After finished, poke the RST button again

srgggaa, 2025-02-25:
just tried and still nothig do i need to do somthing whth an sd card

GuMAZ777, 2025-03-02:
press and hold the decoder before you plug in the usb. i find it easier to remove the battery. unplug it first

srgggaa, 2025-03-02:
I also did that but still nothing the only firmwe that  I can get to work is capibara zero

GuMAZ777, 2025-03-02:
https://youtu.be/GkiQQ8C96mM?si=PLXth9o05LsxkvZ0

Huzzla101, 2025-03-16:
Mr BMorcelli explains it all very well and it must work when you do as he writes... ? 

BUT you can also push the BOOT button,  is simply the button in the middle of the Great Wheel.... Turn it off without the usb plugged then hold that button when you enter the usb then it wont turn on but goes to BOOT MODUS and  if needed check that your pc gives it a com port... It is actually that easy =) 

(if your pc dont give it a com port you can also deactivate then activate it, the usb 3.10 extensible host controller is my name... Yours most likely is about the same name  with processor name in front AMD etc in your   device manager/USB CONTROLLERS,  then it should work , worst case restart...And make sure you using a usb with data transfer not just charger )

Huzzla101, 2025-03-16:
So the BMorcelli way works 100% as thats how its suppose to be done.. BUT since its a T EMBED CC1101 we have the privilege to have a fabric made boot button to besides the RESET button .. YOU DIG ??? =)

srgggaa, 2025-03-16:
That is what I have been doing the whole time and yet it still doesn't work there are more people with the same problem on the same device

Huzzla101, 2025-03-16:
You have the RST/RESET button way that Pirata explains thoroughly that will
work 100% and if you must have the boot way then simply hold the button
inside the wheel when the device is turned off. Then enter the usb while
holding that button down, then it will not turn on but go to BOOT modus,
then pc will give you a com port and you ready to go... If that doesnt work
something is wrong with ur device... I HOPE that you are using a usb with
data transfer.....

srgggaa, 2025-03-20:
I wet over everything like 5 times and even bought a new cable and it does not work (no display)

diynevala, 2025-08-19:
In linux they say you need to also run a command:

`sudo setfacl -m u:USERNAME:rw /dev/ttyACM0`

where USERNAME is your username. But even after running that, I could not get chrome to flash Bruce to Lilygo T-Embed CC1101.

**Instead, I had to boot up Windows 11 and it flashed it at FIRST TRY.**

The only important this is to shutdown first, then hold the center button while plugging in the USB.

I think linux has more than one security issue with the flashing, that is, the security is too tight, and I have no idea how to remedy that.