BruceDevices/firmware #847: 1.9" TFT RST PİN

biscuitaep, 2025-02-26:
Hello i decided to use ESP32S3 to make a custom pcb like smoochiee board i can see TFT RST is connected to a gpio on smoochiee board i dont have enough pin can i connect TFT RST pin to ESP32S3 EN pin?

biscuitaep, 2025-02-26:
@bmorcelli

bmorcelli, 2025-02-26:
> @bmorcelli also do you think should i connect E07 433M20S GDO2 pin to gpio or it can be floating

It depends if you will keep GDO2 set in platformio.ini for this env... If not set, you can ground it with no problem..

I believe RST must be pulled-up, if not used, so you can save this gpio.. got this information on t-Embed CC1101 schematics for st7789 display

biscuitaep, 2025-02-28:
Thanks, did you ever tried using UART0 (Tx0 Rx0) to use gps or something that works on Uart. İm asking because ESP32 S3 works directly with internal USB OTG (D+ D-)

biscuitaep, 2025-03-03:
@bmorcelli Did you try it? Connecting a UART device to TXD0 and RXD0, im just trying to ensure its corrrct before ordering the pcb

bmorcelli, 2025-03-03:
No, I didn't.. and it is not good because you cant debug with serial monitor... Leave these pins for usb only

biscuitaep, 2025-03-03:
> No, I didn't.. and it is not good because you cant debug with serial monitor... Leave these pins for usb only

But i read something that you can redirect Serial0 to USB OTG and debug with it, and use Serial1 for Gpio 43 44 (normallu UART0)

closed 2025-03-28 by bmorcelli