BruceDevices/firmware #850: feature request

tommyfart, 2025-02-28:
In the IR section, can you add something called "Projector-B-Gone" its like "TV-B-Gone" but you can turn off projectors

Lamnxzp, 2025-02-28:
😮‍💨

Lamnxzp, 2025-02-28:
Just use Custom IR with IR Files that you can find on Github (like Flipper-IRDB).
There is no point of adding a "Projector-B-Gone", because first [TV-B-Gone](https://wikipedia.org/wiki/TV-B-Gone) is a real thing and also because esp32 have limited flash and RAM memory.

tommyfart, 2025-02-28:
> Just use Custom IR with IR Files that you can find on Github (like in Flipper-IRDB). There is no point of adding a "Projector-B-Gone", because first [TV-B-Gone](https://wikipedia.org/wiki/TV-B-Gone) is a real thing and also because esp32 have limited flash and RAM memory.

how do i add IR files?

LEONKeenan, 2025-03-04:
you can check out https://github.com/UberGuidoZ/Flipper/tree/main/Infrared , and add the ir files via web portal

closed 2025-03-21 by IncursioHack