BruceDevices/firmware #854: T embed cc1101 all issues I have come across

FTWLuke, 2025-03-01:
Hi, I've seen a lot of people talking about these things and I'm also noticing them. whenever I'm trying to use RF for example scan/copy it's not working. It's not detecting any signals, for custom sub gigahertz It also will not send any of the files I select as well as the spectrum and square wave spectrum. The Jammers also do not work so I think it's just a bug with the entire RF menu. For the RF it does not scan for Mifair ultralight also, if possible, I would like to request that if you hold down the button at the top for more than 5 seconds, it shuts the device off instead of having to go into the menu

Huzzla101, 2025-03-08:
So who is this dude " Alot of people"... Tell him that i am also using a
T-Embed CC1101  and everything works FINE.... Please try to resonate with
the rest of us so it may be possible to work you to find out what your
doing wrong... When i blast the jammer, no more spammer-- :p

rennancockles, 2025-03-08:
I also use t-embed cc1101 and everything's working fine here. Also, to turn the device off, you can hold the top button an turn the wheel counterclockwise

foltz61, 2025-03-11:
I'm new to this but RF doesn't appear to be working for me either.

Huzzla101, 2025-03-12:
Mr *foltz61*  please give a little more details (i am not one of the main
programmers that keeps rocking everday !! =) But i will try help as this is
defined as a "luxury problem" and most likely its just a small detail
missing... You use a Tembed cc1101 or m5stick or.... Did u check CONFIG,
and made sure your pinouts is configured correctly? Tembed have its own
cc1101 module same with m5 sticks so make sure all is correct there ... Do
you have 2 devices so you can send LIVE signals to the recvr to make sure ?
If not then how do you check if all is ok ? We should manage to fix ur
issue easy...   KEEP ON ROCKING IN THE FREE WORLD...

foltz61, 2025-03-12:
I do use the T-embed CC1101. In the RF/Comfig I have CC1101 selected. I haven't messed with RX/TX pins because everything I've seen says changing them isn't necessary if you have chosen your device. When I go to spectrum nothing appears to be detected. I've tried several sets of keys as a test.

Like I said, I'm new to this so it very well could be my fault.

Huzzla101, 2025-03-13:
Nobody have a fault, we are here to explore and develop adventures together
or what =)  But the CC1101 lilygo internal antennas is made for short
distance, when i use the spectrum on that and also i use a m5 stick c plus
2 with external antenna that goes alot further then Lilygo, at the same
time, then lilygo picks up only about 10% of what the m5stick antenna
does.. So it can take some time before it picks up any RF ... Adjust the
RSS to lowest as a start (80% accurate ) in RF meny when u doing a
scan/copy and let it scan for 1hour if you have RF activity it should pick
up something...  As long as it says CC1101 at the top of screen when you
enter RF it should be working ...