BruceDevices/firmware #862: CC1101 don't initialized by  Bruce 1.9

liikata, 2025-03-03:
Use Bruce 1.8 - CC1101 detected.
Use Bruce 1.8.1 - CC1101 detected.
Use Bruce 1.8.2 - CC1101 detected.
Use Bruce 1.9 - CC1101 not detected.

CC1101 don't work with Bruce 1.9
In RF -> RF config -> RF module :
When select (CC1101 share )- CC1101 not found;
When select( CC1101 legacy) - SW stuck- reset for recovery.

Reset of M5 don' fix issue.
Reset of CC1101 don't fix issue.

bmorcelli, 2025-03-03:
I fixed it on BetaRelease.. it'll be on 1.9.1 that will be released asap

liikata, 2025-03-03:
Great - Thanks.

stevedee78, 2025-03-05:
> I fixed it on BetaRelease.. it'll be on 1.9.1 that will be released asap

I now have the CC1101 running with the beta release. Thank you.

BabyMilo01, 2025-03-07:
quick question are any of your subz working because my crashes when i try to use them

liikata, 2025-03-09:
Hi.
1.9.1 successful init CC1101.
Capture function is Ok.
Immediately replay after capture is Ok.
Replay from saved capture is Nok, but this is old issue.

closed 2025-03-21 by IncursioHack