BruceDevices/firmware #865: BC547 transistor substitutions?

bisaw37, 2025-03-03:
I saw the new layout for using the cc1101 with an SD card, and I want to know what other transistors I can use. I have a bunch of S8050 and S8550 transistors laying around, and I want to know if I can use those instead to save me having to buy specifically the BC547.

Also, since I want to integrate this into SuperBruce, a list of usable transistors would be nice, so that I can specify in the coming release what others can be used, in case some people have others on hand.

bisaw37, 2025-03-03:
And also also does the resistor HAVE to be 5k ohm? Or can it be a range of values?

bmorcelli, 2025-03-03:
BC547 is the one that I have tried... the thing is making it work as a NOT port.. I beliew 2N222 and S8050, S8550 might work as well..

the 5k ohm is to polarize it as a switch, you can use 2x10ohm in parallel (to get 5k equivalent), or something smaller, or even connect directly.

give it a try and tell us what you have found

bisaw37, 2025-03-03:
Can I use 4.7k or 5.6k instead? I have single resistors in those values, and it would be simpler to have one instead of 2.

Also by 10 ohm you meant 10k ohm right?

bisaw37, 2025-03-03:
> About the Nrf24, you need to add a switch to cut the 3.3V to Or NRF Or CC1101, not both at the same time

Yes, that is what I was already doing.

> You don't need an extra transistor.. just route it to the same CC1101 pins

I just wanted to make sure I wasn't missing something 😅

Ok thanks for the info!

closed 2025-03-03 by bisaw37