BruceDevices/firmware #866: NRF24 not working on T display S3

doyavivo, 2025-03-04:
I just upload my T display S3 with latest version bruce (1.9). I follow wiring scheme...but not working.
Can someone help?

OriginalNode, 2025-03-31:
I tried this config but same issue, not working,
VCC	3.3V ⚠️
GND	GND
CE	        GPIO 3 
CSN	GPIO 10
SCK	        GPIO 12
MISO	GPIO 13
MOSI	GPIO 11

@bmorcelli  perhaps and idea ?

djaysan, 2025-05-05:
It works...
USE_CC1101_VIA_SPI
#define CC1101_GDO0_PIN 44
#define CC1101_SS_PIN 1
#define CC1101_MOSI_PIN 3
#define CC1101_SCK_PIN 43
#define CC1101_MISO_PIN 2

USE_NRF24_VIA_SPI
#define NRF24_CE_PIN 44
#define NRF24_SS_PIN 1
#define NRF24_MOSI_PIN 3
#define NRF24_SCK_PIN 43
#define NRF24_MISO_PIN 2

bmorcelli, 2025-05-26:
https://github.com/pr3y/Bruce/blob/main/boards/lilygo-t-display-s3/connections.md

closed 2025-05-26 by bmorcelli