BruceDevices/firmware #867: Unable to add multiple custom RFID / MIFARE keys

8ight9ine3three, 2025-03-04:
Hi,

I am trying to add more than 1 custom MIFARE key to the bruce config file (on my SD card)

I have edited the config file to include a long list of custom MIFARE keys, 

but when I restart Bruce w/ the new config file, all the custom keys disappear.

I am adding each key inbetween closed quotations as follows:

"mifareKeys": ["FFFFFFFFFFFF"
"A0A1A2A3A4A5"
"D3F7D3F7D3F7"
"000000000000"
"B0B1B2B3B4B5"],

Any help would be appreciated, I have seen another thread where this topic was addressed and it was noted that it should be possible to add more than 1 custom key / user key dictionary

Thanks

rennancockles, 2025-03-04:
Your list format is wrong. It is missing a comma ',' between the elements of the list. That's why yours keys are disappearing

8ight9ine3three, 2025-03-05:
Thank you @rennancockles, that was the solution. Rookie mistake on my end

closed 2025-03-05 by 8ight9ine3three