BruceDevices/firmware #883: Sd card

bob6969696900, 2025-03-06:
When i go onto files ad try click on sd card it says the folders but when i open them it goes blank then reboots im on the newest firmware for lilygo t embed cc1101 ?

johnalpha911, 2025-03-27:
i have the same problem

bmorcelli, 2025-03-28:
can you test the last Beta, please?

johnalpha911, 2025-03-28:
Im in beta right now and it has a bug i can't type !@#$%^&*() for some reason or any symbol that you have to press control and the letter or number to type

johnalpha911, 2025-03-28:
Because i use the cardputer

jphphotography, 2025-03-31:
I just figured out a fix for my issue, yours might be related, make sure there is a folder called "BruceRF" in the root of your SD card. FYI I figured this out by monitoring the the output from the serial port during the crash.

johnalpha911, 2025-03-31:
I actually now use an other SD card with no problems

blueskiesoky, 2025-04-06:
When I try to open subfolders, it only gives option to new folder, delete folders. It does not browse and display files

closed 2025-05-26 by bmorcelli