BruceDevices/firmware #885: hi i have bruce 1.9.1 on my m5stickc plus 2 and im experiencing an issue with the custom subz. when i try to use any of them it forces my device to reboot, is there anything to help fix it

BabyMilo01, 2025-03-07:
**Describe the bug:**

**Steps To Reproduce:**

- Device used: 

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

SamuEDL, 2025-03-09:
It also happens on the T-embed C1101 with bruce 1.9.1, that you try to start custom SBz and it causes it to reboot.

Device used: Lilygo T-embed C1101 (M5 launcher 2.4.6 and Bruce 1.9.1)

IncursioHack, 2025-03-27:
There have been several changes made to RF in the last few weeks, please test it in BETA at https://bruce.computer.

jphphotography, 2025-03-31:
While writing this I found the issue, I'm still keeping everything below just to be thorough but in my instance at least, the issue stemmed from not having a BruceRF folder present on the SD card on my Cardputer. The reason it worked on my CYD is because I had used the RX to capture a file first and that must have created the folder already. 

*****************************************************
I'm facing this issue on the carputer as well, running Bruce 1.9.1 (from M5 Launcher) oddly no issue on my CYD flashed via the webflasher. 

Describe the bug: Trying to open the file browser for SD Card to choose a Custom SubGhz file from with the RF section causes unit to immediately reboot. Appears to also do the same thing if selecting LittleFS. To be clear it does not get to the point where you can see any files it crashes immediately after choosing SD Card. 

Steps To Reproduce: On a cardputer running Bruce 1.9.1 loaded via M5 Launcher, within RF>Custom SubGHz select SD Card as the source. Does not require a module to be present, seemed

Device used: Cardputer (crashes when opening filebrowser for SD Card and LittleFS to find custom subghz file). Note I also have a CYD(2.8" Resistive touch screen with 1 micro usb, flashed using inverted colours option selected), on this unit I can browse files on my SD card without any issues. On the CYD it crashes when I choose LittleFS though - Full disclosure I'm not super familiar with the LittleFS function, crash might be due to not having SPIFFs?
Expected behavior: File browser would open and allow you to look through files and choose one. 

Actual behavior: On cardputer it crashes upon trying to open the filebrowser for SD Card and LittleFS within the customer subghz menu option. On CYD it works for SD card but not LittleFS.

Additional context / logs:
I think there is at least 1 other issues that is either directly related or the same exact issue 
https://github.com/pr3y/Bruce/issues/883

Serial dump from Cardputer during crash
[1090535][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Usin "cs": 1,
    "io0": 2,
    "io2": -1
  },
  "NRF24_Pins": {
    "sck": 40,
    "miso": 39,
    "mosi": 14,
    "cs": 1,
    "io0": 2
  },
  "SDCard_Pins": {
    "sck": 40,
    "miso": 39,
    "mosi": 14,
    "cs": 12,
    "io0": -1
  }
}g Channel 7, Value = 160
Selected: Custom SubGhz
Selected: SD Card
[1105602][E][vfs_api.cpp:105] open(): /sd/BruceRF does not exist, no permits for creation
[1105627][E][vfs_api.cpp:105] open(): /sd/BruceRF does not exist, no permits for creation
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42269ebf  PS      : 0x00060c30  A0      : 0x8200e7d0  A1      : 0x3fceb790
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000008  A5      : 0x3fceb7bc
A6      : 0x00000000  A7      : 0x0000a80f  A8      : 0x8212c1d4  A9      : 0x3fceb740
A10     : 0x00000000  A11     : 0x00000002  A12     : 0x60025000  A13     : 0x00000003
A14     : 0x00000004  A15     : 0x3fca4100  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000f  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff

Backtrace: 0x42269ebc:0x3fceb790 0x4200e7cd:0x3fceb7b0 0x4202652e:0x3fceb800 0x4205f4f9:0x3fcebc10 0x42016d1f:0x3fcebdc0 0x4200e69a:0x3fcebde0 0x4200f601:0x3fcebe00 0x420171be:0x3fcebe70 0x4226a40d:0x3fcec140 0x420365d5:0x3fcec160 0x4213279c:0x3fcec210

ELF file SHA256: ebc66b28d2623f12

Rebooting...

IncursioHack, 2025-03-31:
> While writing this I found the issue, I'm still keeping everything below just to be thorough but in my instance at least, the issue stemmed from not having a BruceRF folder present on the SD card on [...]

Please try some of these actions.:

1 - Please install the BETA version, as shown in the image at:
https://bruce.computer/flasher

2 - If you want the Binary to insert into your SD and run with the Launcher, you can download it here:
https://github.com/pr3y/Bruce/tree/WebPage/BetaRelease
OR
Lilygo T-Embed CC1101: https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-lilygo-t-embed-cc1101.bin
Cardputer: https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-m5stack-cardputer.bin
StickC Plus 2: https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-m5stack-cplus2.bin

3 - Backup and Deletion of bruce.conf Files

• Backup: Before deleting anything, make sure to back up any important data. You can back up the files directly from the WebUI.
Delete bruce.conf files:
• Access the WebUI of the Bruce from another device (computer, laptop, or smartphone).
• Go to the LittleFS menu.
• Locate and delete the bruce.conf files from LittleFS (internal storage).
• Go to the SDCard menu (if present).
• Locate and delete the bruce.conf files from the SD Card (external storage), if present.
• After deletion: Restart the Bruce and test the device again to check if the problem persists.