BruceDevices/firmware #888: sd card on t-embed cc1101

arbubyaka, 2025-03-07:
Describe the bug:
The T-Embed CC1101 device does not recognize a 32 GB SD card formatted in FAT32, while a 128 MB SD card (also formatted in FAT32) works correctly.

Steps To Reproduce:
Device: T-Embed CC1101.
Insert a 32 GB SD card formatted in FAT32.
Attempt to read/write data via the device interface.
Repeat the steps with a 128 MB SD card formatted in FAT32.

Expected behavior:
Both SD cards (32 GB and 128 MB) should be detected by the device and support FAT32 read/write operations.
Actual behavior:
The 32 GB SD card is not detected by the device or triggers an error.
The 128 MB SD card works without issues.

Additional context / logs:
Possible causes:
The device has a maximum supported SD card capacity limitation.
Errors in the SD card driver implementation within the device.

Verified actions:
The 32 GB SD card is successfully detected by other devices (PC).

Formatting was performed via Rufus with FAT32 settings (32 KB cluster size).

Workaround:
Use SD cards with a capacity of ≤ 2 GB (SDSC) for compatibility with the T-Embed CC1101.

Wishes:
[image]

Huzzla101, 2025-03-08:
Try to do a normal clean windows formatting maybe . I am using T-Embed
CC1101  and a 32gb SD and it works FINE... =)  I even tried with a SD card
with no brand and it worked also..

blueskiesoky, 2025-04-08:
On mine, it will show folders, but will browse into them. Everything in root folder and BruceRF folder directly shows ir and sub files. When selecting other folders, it just displays options to rename, new folder, delete folders.

Pryxc, 2025-10-31:
The problem is back,When you change the SD card, the device sees it, restarting the device kills the SD

SD Card "Kingston" 32G FAT-32 

Device: T-Embed cc1101