BruceDevices/firmware #892: T Embed CC1101 RF is kinda flimsy

mxrxtr, 2025-03-08:
Describe the bug:
The Device doesnt catch the rf signal most of the time even if held close together

Steps To Reproduce:
Go to RF then scan copy set a specific range  set Treshold to -80 

Expected behavior:
The Signal gets reiceved every time

Actual behavior:
Catches the signal only in rare cases

Additional context / logs:
Tested different sources and made sure im on correct frequenz

mileshkin, 2025-03-08:
This is the correct behavior.
The range modes are designed to determine what frequency the device you want to scan operates on.
The sensitivity is reduced specifically so that you can only catch the signal of a source nearby, very close, and not the entire RF spectrum that surrounds us.
After you have determined the frequency the device operates on, simply select "Select Fixed" and the reception is already very good in this mode.
The range mode should be perceived as a near-field frequency meter, and the fixed frequency mode as the main working one. You can additionally make an external antenna modification.

mxrxtr, 2025-03-08:
@mileshkin thats not what i meamt im on tje correct frequenz and it dpesnt catch it

mileshkin, 2025-03-08:
While range is set to some xxx-yyy it stays in attenuated frequency meter mode.
Get the exact frequency of your remote and set it in "fixed" then scan again - everything is going to be good that time.

closed 2025-03-10 by mxrxtr