BruceDevices/firmware #893: Wardriving not escaping SSID names

henriquetorquato, 2025-03-08:
**Describe the bug:**
During wardriving, I found a WIFI network where the SSID was prefixed as "2,4 GHz". This breaks the CSV formatting and visualization since the text after the comma is considered to be a new field.

**Steps To Reproduce:**
- Under GPS > Wardriving
- Detect a SSID with a comma
- Open CSV in LibreOffice
- Columns are broken
- Device used: M5's Cardputer, Bruce v1.9

**Expected behavior:**
SSID names should be escaped since the separator character can be on the content.

**Actual behavior:**
SSID name is not escaped and when a comma is present on the name, it is broken into two columns.

**Additional context / logs:**
A simple double quote escape on this line should be enough: https://github.com/pr3y/Bruce/blob/bce2af94fd6f9e551839702cd560a3a7fdf8336e/src/modules/gps/wardriving.cpp#L230

I double checked WiGLE's documentation on the WigleWifi-1.6 format and they state "The WiGLE CSV format adheres to UTF-8 encoded [rfc4180](https://tools.ietf.org/html/rfc4180) for the most part". I checked rfc4180 and on section 2 topic 5 it states "Each field may or may not be enclosed in double quotes ... If fields are not enclosed with double quotes, then double quotes may not appear inside the fields."

That means that double quoting the SSID field should be supported on WiGLE's file importer, and that maybe an additional double quote character parsing should also (maybe) happen.

Honestly, I'm not even sure if a double quote is a supported character for SSID names.

henriquetorquato, 2025-03-08:
@rennancockles I opened a PR with a fix for the issue.
Is there a testing framework in place for the project? I would like to add at least unit tests for this method.

rennancockles, 2025-03-08:
No, we don't have anything yet.

henriquetorquato, 2025-03-08:
@rennancockles is it ok if I use [Microsoft's C++ Framework](https://learn.microsoft.com/en-us/visualstudio/test/microsoft-visualstudio-testtools-cppunittestframework-api-reference?view=vs-2022)?

I think it's fairly compatible to how other popular testing frameworks works.

I can also leave this out for now to have this decided not on a issue.

rennancockles, 2025-03-08:
You would have to check with @bmorcelli . He's the one who maintains the pipeline

bmorcelli, 2025-03-08:
Pipeline runs on Ubuntu machines..

I believe that Platformio test builds can be used somehow, using the Arduino-ESP32 framework used in bruce project, so as rennan said, it wasn't implemented yet

henriquetorquato, 2025-03-08:
@bmorcelli @rennancockles I'll give it a go using GoogleTest. Looking at the docs, it seems to be the one with the most compatibility out of the supported frameworks: https://docs.platformio.org/en/stable/advanced/unit-testing/frameworks/index.html#testing-frameworks

If the setup gets too big I'll drop it to be done another time.

henriquetorquato, 2025-03-09:
@bmorcelli I'm having a little trouble with the test setup, specially mocking the Arduino framework code. I'm also not sure about the configuration of the tests either, since it seems that they can also run on a hardware level.

I'll drop the implementation for now, will do a test build and will try it out next week (although the change itself seems safe enough to be merged without testing).

I think for the future it should be considered 2 separated levels of tests:.
- [Native](https://docs.platformio.org/en/latest/advanced/unit-testing/runner.html#native) tests for unit testing
    - To be executed locally or on the pipeline
- [Embedded](https://docs.platformio.org/en/latest/advanced/unit-testing/runner.html#embedded) tests for hardware e2e integration tests
    - These might not be desired for the project though

closed 2025-03-10 by henriquetorquato