BruceDevices/firmware #896: Request: autosave RF signals

asturale, 2025-03-08:
**Is your feature request related to a problem? Please describe.**
I'd like to keep RF scan/copy running for a while and capture and save detected signals without user interaction to analyze them later.

**Describe the solution you'd like**
Add a menu option to enable autosave. 
When a signal is detected (and shown on the display) and autosave is enabled, save the signal .sub to SD card with the timestamp as filename.

**Describe alternatives you've considered**
Flipper Zero does this for decoded signals but not for raw.
RTL-SDR does this but is less portable and can't replay.

tegtheevi, 2025-04-02:
This would be a great idea 
similar to evilcrow v2 but in automode

tegtheevi, 2025-04-08:
https://github.com/GthiN89/NucleusGateKeeper 
This repository can help you for now i built it and it works like magic

rennancockles, 2025-04-08:
Already added to beta version